neXenio/BLE-Indoor-Positioning #50: Current Location Accuracy

tekinalper, 2018-03-28:
Hi, firstly thank you so much for your effort for such a good project. We all have chance to see the implementations of several calculations and algorithms all together.

I have pretty tested the library and app in a real office environment with the beacons installed. Here I have some test results and comments;

Although it was quite difficult to determine the coordinates of the beacons, In the first scenario I have tested with TxPower=6 which is maximum range advertising ~50 m. Each beacon has 7 m distance and 12 beacons installed in total. Mostly showing the right location when I move but sometimes current location dot moves out of the region or to other corner etc. I assumed maybe it shows wrong location on the view but also checked the locations on multilateration result and saw some deviations. Another point that caught my attention is possible wrong calculation of edge locations. 

In the second scenario, I have tested by setting TxPower=1 which supposed to advertise in 4 m range but test is not really successful maybe because of some other needed configurations. Also no good result when I played with deviceAdvertisingRange parameter in BeaconMap(Even the same when I set 50 m for the first scenario).  

However, I see that you convert cartesian coordinates to spherical coordinates. Do you confirm that the convert is totally gives the same successful location calculation according to cartesian?

Additionally, I appreciate if you have any other configuration suggestions?

Thank you,
Alper

leonlowitzki, 2018-03-29:
Hi Alper,
we are happy to hear you are testing our project and really appreciate your feedback! We started to write a [wiki](https://github.com/neXenio/BLE-Indoor-Positioning/wiki/RSSI-Measurements) for more documentation. For the conversion from cartesian coordinates to spherical coordinates, we wrote [tests](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/test/java/com/nexenio/bleindoorpositioning/location/projection/SphericalMercatorProjectionTest.java). Since we are still working on this, you could really help us by answering some questions:
- How / in what shape did you arrange the beacons?
- Did you use the same distances between beacons for both scenarios?
- Can you explain in what way the second scenario performed worse i.e. unstable location?
- Could record the screen of the app?
- Can you provide more detail about the wrong calculation of the edge locations?
Thank you!
Leon

tekinalper, 2018-03-29:
Hi Leon,

 Thank you for your answer. Nice to see the wiki, I will be waiting for further documentation. 

I see your tests, actually the key point is the difficulty of determining the beacons' geo location coordinates in the indoor area. 

- Blue pins are the beacons. Currently, I placed each beacon on a high place with a clear vision. Later, will test after deploying them to the ceiling. 
[image]
- I had changed TxPower from 0 dBm to -20 dBm but it is my bad, forgot to change calibrated power. Now set to -79 dBm and have better result but still less accurate. Also it solved the "wrong calculation of the edge locations" hypothesis.
- [Here](https://drive.google.com/open?id=1fJcaJvGojDGfmPh2YPWtbmt8ssCk709g) is the screen record. My location is almost at the center.
- I used same distances for both scenarios. Also trying to decide which gives more accurate result; when TxPower is 0 or -20 dBm. Does it make any difference, what do you think?

In conclusion: 
- 12 beacons, approximately 7 meters distance between beacons.
- TxPower: -20 dBm and cPwr: -79 dBm
- IndoorPositioning / UPDATE_INTERVAL_MEDIUM
- BeaconMap / deviceAdvertisingRange = 4 m
- We have more than 100 beacon devices in the office. Tests might being affected by signal noise etc. 

Looking forward to your comments!

leonlowitzki, 2018-04-03:
Hi Alper,

thanks for the feedback! We will be writing a wiki post, when we have properly documented how to get the exact geo location of beacons in an indoor environment. Have you considered changing  the`updateLocation` method in the [Indoor Positioning Class](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/IndoorPositioning.java)? You could select the beacons with the lowest RSSI values for example, to only use the beacons with the best signal quality.

There is a lot of research considering the „best“ TxPower setting. Generally speaking a bigger advertising range makes more beacons available for any given position. On the other hand a larger distance between beacons increases the chance of signal attenuation in form of walls or other sources of error. Some researchers even evaluate the possibility of using different, individual TxPower settings for each beacon.

Thanks,
Leon

leonlowitzki, 2018-04-13:
Hi Alper,
here is the promised wiki post https://github.com/neXenio/BLE-Indoor-Positioning/wiki/Indoor-Coordinates. Looking forward to you feedback!

Steppschuh, 2018-04-17:
Hey Alper,
as promised by Leon, we've created a wiki post that explains how to get the geo-coordinates of beacons. You can find it [here](https://github.com/neXenio/BLE-Indoor-Positioning/wiki/Calculating-Geo-Coordinates-of-Beacons).

You should try to deploy our current `dev` branch, the locations should be more stable now, even with many beacons around. Depending on you use-case, you can also use the mean location instead of the latest multilateration result (see #55)

Also, I've seen that you made some adjustments (e.g. the beacon labels in the map view). If you think that your changes could improve this project, feel free to submit pull requests.

Steppschuh, 2018-10-29:
Hey Alper,
I'm closing this due to inactivity. You can checkout the current beacon map with a background image here: #90

closed 2018-10-29 by Steppschuh