neXenio/BLE-Indoor-Positioning #71: Location Accuracy Issue

hadiidbouk, 2018-06-05:
Hello,

Thank you, for supplying us with this open source indoor position solution.

We have been working on your solution for a few days and were able to reach the users locations using the beacons; however, there seems to be some accuracy issues, as we are getting many location responses that are varying in location even though we are not moving.

We are currently using 4 beacons supplied by Kontakt, (2 [Pro](https://store.kontakt.io/next-generation/32-beacon-pro.html) and 2 [standard](https://store.kontakt.io/our-products/30-double-battery-beacon.html)). Each beacon has been set with a TX power of -8dBm.

Also note that we have set the scanning mode of the BLE to `SCAN_MODE_LOW_POWER`.

See below image, which is a screenshot of our region surrounded by beacons (A, B, C, and D) which are mounted on columns at a height of about 2.5m, also note that the items in the middle are benches found in the area and not columns that might possibly interfere with the beacons signals. 

[image]

This is a screenshot showing the right recordings for our location by the beacons.

[image]

We have also attached a [video](https://drive.google.com/file/d/1MwCFq8RUHQimgNKlorUtO7QDaYr3sdim/view?usp=sharing) screen recording of our sample application showing the location error, and dots (our location) vastly varying in the whole map.

 Thank you in advance for your help.

leonlowitzki, 2018-06-05:
Have you tried using our [sample app](https://github.com/neXenio/BLE-Indoor-Positioning/tree/master/app)? You can set your beacon locations in the [bluetooth client class](https://github.com/neXenio/BLE-Indoor-Positioning/blob/becd71450afa051c2c54a967b1d44eb76eeb538d/app/src/main/java/com/nexenio/bleindoorpositioningdemo/bluetooth/BluetoothClient.java). Other sources of error might be inaccurate 1 meter rssi calibration or a different environment requiring a different path loss parameter. You can find both in the [beacon distance calculator class](https://github.com/neXenio/BLE-Indoor-Positioning/blob/bb48bd83cc42b1bc9816e26eaaaf33f94cc6874a/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/location/distance/BeaconDistanceCalculator.java#L61). You can find your rssi at 1 meter like [this](https://support.kontakt.io/hc/en-gb/articles/115000115610-Reference-RSSI-Value-Calibration) and different path loss exponents for different environments [here](https://en.wikipedia.org/w/index.php?title=Log-distance_path_loss_model). You can also try using [getMeanLocation](https://github.com/neXenio/BLE-Indoor-Positioning/blob/b1dbfc18ea7a94608199596ae5245fa08d40b5ba/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/IndoorPositioning.java#L66) in the dev branch to stabilize your location.

Steppschuh, 2018-06-06:
For reference: here's a GIF showing the accuracy that we get using the sample app in our office environment.

[image]

Our beacons are mounted to the ceiling and have exact geo coordinates, as explained in [our wiki](https://github.com/neXenio/BLE-Indoor-Positioning/wiki/Calculating-Geo-Coordinates-of-Beacons).

hadiidbouk, 2018-06-07:
Thank you guys for the fast reply,

I think that the problem was with our beacons location, now we have 12 digits after the decimal point for each longitude and latitude and that give us a better result.

Also i am using now the Bluetooth Client class instead of my custom implementation.

However, the marker is never get close to the beacon attached on the column, i don't know why.

Best

Steppschuh, 2018-06-08:
We also want to provide more details about the multilateration results, so that you can request the accuracy / variance / standard deviation of the currently calculated locations. See #74

closed 2018-06-08 by Steppschuh