neXenio/BLE-Indoor-Positioning #79: meetsSpecification not working with all ibeacons

hadiidbouk, 2018-06-07:
Hello,

I am currently using 4 beacons supplied by Kontakt, (2 [Pro](https://store.kontakt.io/next-generation/32-beacon-pro.html) and 2 [standard](https://store.kontakt.io/our-products/30-double-battery-beacon.html)).

The beacons (Pro), doesn't meet the specification of `IBeaconAdvertisingPacket` class.

I am getting different values of `EXPECTED_TYPE` and `EXPECTED_BEACON_TYPE` for the pro type of beacon. I don't know if Kontakt change something in the pro beacons.

I have test this with (aruba, estimote, averos, standard kontakt) and everyone works fine except the Kontakt pro beacon.

``` 
 public static boolean meetsSpecification(byte[] data) {
        if (data == null || data.length < 29) {
            return false;
        }
        if (getTypeBytes(data) != EXPECTED_TYPE) {
            return false;
        }
        if (!Arrays.equals(getFlagsBytes(data), EXPECTED_FLAGS)) {
            return false;
        }
        if (!Arrays.equals(getBeaconTypeBytes(data), EXPECTED_BEACON_TYPE)) {
            return false;
        }
        return true;
    }
```
Thank you

Steppschuh, 2018-06-08:
The `meetsSpecification` method is based on the specification of the iBeacon advertising packet structure. According to [Kontakt.io](https://support.kontakt.io/hc/en-gb/articles/201492492-iBeacon-advertising-packet-structure) their packets meet the specification, at least if they use the specified default values.

`EXPECTED_TYPE ` checks byte 4, which indicates the advertising data type and defaults to `0xFF`. However, it may be customized by the beacon manufacturer. I agree that we shouldn't expect this value in our check, it will be fixed in an upcoming PR.

`EXPECTED_BEACON_TYPE` checks byte 7 and 8, which are required to have the value `0x02` (iBeacon) and `0x15` (length), as per specification. I don't feel good about adjusting this check, as all iBeacons should advertise these exact values.

Could you post the byte array that you get from your Kontakt Pro beacon?

hadiidbouk, 2018-06-11:
Thank you for your reply,

Here is the the byte array from the Kontakt Pro Beacon

You can see that the byte 4 is 22

```
0 = 2
1 = 1
2 = 6
3 = 13
4 = 22
5 = 106
6 = -2
7 = 2
8 = 6
9 = 1
10 = 10
11 = 82
12 = 4
13 = 98
14 = 78
15 = 101
16 = 53
17 = 5
18 = 9
19 = 98
20 = 78
21 = 101
22 = 53
23 = 0
24 = 0
25 = 0
26 = 0
27 = 0
28 = 0
29 = 0
30 = 0
31 = 0
32 = 0
33 = 0
34 = 0
35 = 0
36 = 0
37 = 0
38 = 0
39 = 0
40 = 0
41 = 0
42 = 0
43 = 0
44 = 0
45 = 0
46 = 0
47 = 0
48 = 0
49 = 0
50 = 0
51 = 0
52 = 0
53 = 0
54 = 0
55 = 0
56 = 0
57 = 0
58 = 0
59 = 0
60 = 0
61 = 0
```

Steppschuh, 2018-06-11:
I've adjusted the check in PR #83 to tolerate changes in byte 4 (`EXPECTED_TYPE`). Make sure to change your dependency to `compile 'com.github.neXenio:BLE-Indoor-Positioning:dev-SNAPSHOT'` in order to get the changes.

I wouldn't consider the advertising data from your Kontakt Pro beacon to be an iBeacon frame. According to Apple's specification:

| Bytes     | Description |
| --------: | -------- |
| 09-24      | Proximity UUID     |
| 25-26      | Major     |
| 27-28      | Minor     |
| 29      | Signal Power     |

However, you can create your own classes for your use case,  e.g. `KontaktProAdvertisingPacket extends AdvertisingPacket` and `KontaktProBeacon<P extends KontaktProAdvertisingPacket> extends Beacon<P>`. Have a look at the `AdvertisingPacketFactoryTest` and `BeaconFactoryTest` to see how you can use your own classes.

closed 2018-06-11 by Steppschuh

hadiidbouk, 2018-06-11:
Please note that the byte 7 and 8 is {2, 6}  and not like the expected beacon type ( in case you want to ignore checking these bytes in the development environment like you did in your PR #83).
 
`EXPECTED_BEACON_TYPE = {0x02, 0x15};`

Steppschuh, 2018-06-11:
Yes, I did notice that. The difference is the length of the advertising data from the Kontakt Pro beacon, which makes it not compliant with the iBeacon specification (see the table from my last comment).

PR #83 (2018-06-11): 0079 ibeacon meets specification
Closes #79