neXenio/BLE-Indoor-Positioning #81: Get the lastknownLocation if the device isn't moving

hadiidbouk, 2018-06-07:
Hello, 

It will be better if the location still the same if i am not moving around, a quick solution to  add `setMinimumMovementSpeed` method.

Steppschuh, 2018-06-08:
You can set your desired maximum movement speed using the `IndoorPositioning` singleton:

```java
double movementSpeed = 5.0; // meters per second
IndoorPositioning.getInstance().setMaximumMovementSpeed(movementSpeed);
```

Keep in mind that you can also use this convenience method `IndoorPositioning.getMeanLocation(10, TimeUnit.SECONDS)` to get more stable results. You can also directly use `LocationUtil.calculateMeanLocation(List<Location> locationList)` if you want to specify your own locations.

hadiidbouk, 2018-06-11:
I have played with these functions yes,
but the problem is why should i receive a new location even if the phone is on top of the table and isn't moving, why should we let the device scanning and making calculation all the time when i am not moving around ?

Steppschuh, 2018-06-11:
The GIF I posted in #71 shows the location changes while the device wasn't moving. The beacon RSSIs are changing rapidly (no matter if the device is moving or not), and we need to process all the values anyway (to keep the signal filters accurate). The multilateration results will be slightly different in each calculation and location listeners should be updated with each result.

It's in your hands to assign a meaning to these locations. If you know that the device hasn't been moving for 10 seconds, you might as well use the mean location from the past 10 seconds instead of the latest multilateration result.

closed 2018-06-11 by hadiidbouk