neXenio/BLE-Indoor-Positioning #84: Unable to plot points

parvinder06, 2018-06-11:
I am not able to get locations.

Only Getting the following values :
AndroidLocationProvider: Received location result with 1 locations
    Last known location set to : Latitude: 30.722795 Longitude: 76.7677966

parvinder06, 2018-06-11:
No

Steppschuh, 2018-06-11:
Please try to report some more detailed issues. We have no idea what you are trying to accomplish, what exactly is causing issues and how you tried to solve them.

`SensorManager` could cause issues if your android device has no accelerometer or magnetic field sensor (or if you blocked access to the device sensors).

parvinder06, 2018-06-11:
Till now i have just compiled this code and got the following result:
[image]
[image]
[image]

hadiidbouk, 2018-06-11:
I get this bug once when i set the Location provider as an iBeacon and the bluetooth scanner detect and Eddystone beacon.

`Java.lang.ClassCastException: com.nexenio.bleindoorpositioning.ble.beacon.Eddystone cannot be cast to com.nexenio.bleindoorpositioning.ble.beacon.IBeacon`

Steppschuh, 2018-06-11:
@parvinder06 what you are seeing is the expected result, if you have not setup your own beacons with their respective location providers yet.

I guess you haven't calculated the geo-coordinates of your beacons yet. Please follow [these steps](https://github.com/neXenio/BLE-Indoor-Positioning/wiki/Calculating-Geo-Coordinates-of-Beacons) in order to do so.

Steppschuh, 2018-06-11:
@hadiidbouk some beacons are capable of advertising multiple frames simultaneously (e.g. iBeacon and Eddystone frames). Are stacktrace would be nice if you think that this is an issue with the library.

hadiidbouk, 2018-06-11:
Sure i will create an issue later

parvinder06, 2018-06-12:
Can you please provide your .apk file ?

hadiidbouk, 2018-06-12:
@parvinder06  I am just running the [sample application](https://github.com/neXenio/BLE-Indoor-Positioning/tree/master/app) now

parvinder06, 2018-06-12:
@hadiidbouk can you please attach .apk file? 
Actually, I want to relate it with my .apk file.

Steppschuh, 2018-06-12:
@parvinder06 this is a debug build APK from the current dev ([e6c387d](https://github.com/neXenio/BLE-Indoor-Positioning/tree/e6c387daa0592926ce23fa1df46c70553649f0b4)): 
[ble-indoor-positioning-dev-e6c387d.apk.zip](https://github.com/neXenio/BLE-Indoor-Positioning/files/2092978/ble-indoor-positioning-dev-e6c387d.apk.zip)

parvinder06, 2018-06-12:
@Steppschuh   I am sorry but the APK you have sent is not getting installed as it is in debug build. You have to make it signed first.

Steppschuh, 2018-06-12:
There's absolutely no reason to sign the APK. You can easily checkout the repo and build (and sign) the APK yourself.

Steppschuh, 2018-06-19:
Closing this because I don't see any issues here. Feel free to re-open.

closed 2018-06-19 by Steppschuh