neXenio/BLE-Indoor-Positioning #86: Sample app crash when using LocationProvider instead of minor

hadiidbouk, 2018-06-11:
Hello,
if someone use the `LocationProvider` to get the location from api or based on the mac address ( and not from the minor ) the sample app will crash when pressing the chart button because he may get a negative value of minor and he cannot get a color from the materialDesignXColors array.

```
 @ColorInt
    public static int getBeaconColor(Beacon beacon, @ColorUtil.ColoringMode int coloringMode, int beaconIndex) {
        int colorIndex = 0;
        switch (coloringMode) {
            case ColorUtil.COLORING_MODE_INSTANCES: {
                if (beacon instanceof IBeacon) {
                    colorIndex = ((IBeacon) beacon).getMinor();
                } else {
                    colorIndex = beaconIndex;
                }
                break;
            }
...
```

Steppschuh, 2018-06-11:
That sounds quite strange.

The `BeaconView` will only use the minor if the specified beacon is an instance of `IBeacon`. The major and minor must be values between 0 and 65535 and thus can never be negative. If they are, please let me know how to reproduce the issue.

Besides that, even if the value of `colorIndex` is negative, that shouldn't be an issue (as we are using the modulo operator).

``` java
colorIndex %= MATERIAL_DESIGN_COLOR_RESOURCE_IDS.length;
return getInstance().materialDesignDarkColors[colorIndex];
```

hadiidbouk, 2018-06-11:
You can see below that the minor is a negative value when create a beacon from 
`BeaconManager.getBeacon(macAddress, advertisingPacket.getBeaconClass());`

[image]

And you can follow this scenario to check how the app crash : 

1 - 
[image]

2 - 
[image]

3 - 
[image]

4 - 
[image]

It seems that there is something not finished yet in background, because when waiting a while everything works fine.

Steppschuh, 2018-06-11:
Could you let me know the raw byte array of advertising data from your beacon? I'll add it as a test case to figure out what could be causing the negative minor.

I also just realised that in Java, the `%` operator is not modulo, as I expected, but instead returns the remainder (according to [StackOverflow](https://stackoverflow.com/questions/5385024/mod-in-java-produces-negative-numbers)). I'll add a fix for that.

hadiidbouk, 2018-06-11:
Sure here you go, it's a kontakt beacon but not the pro version.

```
0 = 2
1 = 1
2 = 6
3 = 26
4 = -1
5 = 76
6 = 0
7 = 2
8 = 21
9 = -111
10 = 20
11 = -42
12 = 26
13 = 103
14 = -47
15 = 17
16 = -24
17 = -83
18 = -64
19 = -6
20 = 122
21 = -32
22 = 27
23 = -66
24 = -68
25 = 100
26 = -16
27 = -113
28 = 90
29 = -59
30 = 8
31 = 9
32 = 75
33 = 111
34 = 110
35 = 116
36 = 97
37 = 107
38 = 116
39 = 2
40 = 10
41 = 4
42 = 10
43 = 22
44 = 13
45 = -48
46 = 119
47 = 99
48 = 74
49 = 111
50 = 52
51 = 50
52 = 100
53 = 0
54 = 0
55 = 0
56 = 0
57 = 0
58 = 0
59 = 0
60 = 0
61 = 0
```

Java sucks.

Steppschuh, 2018-06-11:
Based on your data and the current implementation, I get the following when parsing as an iBeacon advertising packet:

`IBeacon Advertising Packet (Proximity UUID: 9114d61a-67d1-11e8-adc0-fa7ae01bbebc Major: 25840 Minor: -28838 RSSI at 1m: -59)`

Could you let me know which values you actually configured for that beacon?

hadiidbouk, 2018-06-11:
Sure here you go

```
Proximity UUID: 9114d61a-67d1-11e8-adc0-fa7ae01bbebc
Major: 25840
Minor: 36698
TX Power: 4dBm
Advertising Interval: 625
```

Steppschuh, 2018-06-11:
This should be fixed in PR #88. I've added your advertising data to the [IBeaconTest](https://github.com/neXenio/BLE-Indoor-Positioning/blob/6351b0cbf4e9c1746449e835d8784e792f15fccb/BLE%20Indoor%20Positioning/src/test/java/com/nexenio/bleindoorpositioning/ble/beacon/IBeaconTest.java).

PR #88 (2018-06-11): 0086 negative minor ibeacon
Fixes #86 and #87
commit cc3fe0d8 (2018-06-11): Adjusted parsing of unsigned ints from 2 bytes, fixes #86
closed 2018-06-11 by Steppschuh