neXenio/BLE-Indoor-Positioning #89: guide for the app

raitani, 2018-06-18:
i installed app on my android phone . can you please suggest how it will work? give a guideP?

Steppschuh, 2018-06-18:
Completely agree that we need a guide for setting up the app with your own beacons / locations / maps. We'll create a post in our [wiki](https://github.com/neXenio/BLE-Indoor-Positioning/wiki) about that in the near future.

raitani, 2018-06-18:
Can you please suggest me how to use it?

ZaxLofful, 2018-07-07:
I am really interested in this getting started guide myself. I just found this while doing a bit of research for positioning, here are the things I would love to see as I am a total newbie at the moment:

1. What modules can I use with this? A link to an example would be great, I am looking at a few on Amazon now.
2. What do I have to do to get them setup?
3. Does the app do all the hard work or do I need to deploy a server?

Steppschuh, 2018-07-09:
Glad that you're interested in this topic. However, I'm not sure if you understand what this repository is providing.

It's a Java library that allows you to calculate geo-coordinates based on the signal strength of nearby Bluetooth beacons. It also contains a demo app that uses this library to visualize its capabilities.

You are in charge of setting up the Bluetooth beacons and assigning geo-coordinates to them. If you need a server to do so, that's also on you to implement.

ZaxLofful, 2018-07-09:
I completely understand that this repo is for, I was merely suggesting a way to up level and get more people interested in your code. I randomly stumbled upon this, but have no idea what beacons to get or how to start attempting this....Even just a link to another GitHub would be cool.

Rahesh31, 2021-01-19:
Hello I want to know whether anyother library like https://altbeacon.github.io/android-beacon-library/index.html should be added to interact with beacon through app or your library does that thing by default ?