neXenio/BLE-Indoor-Positioning #93: registerLocationListener method parameters convention

hadiidbouk, 2018-06-18:
Hello,

The registerLocationListener method in `AndroidLocationProvider` are not nullable

`registerLocationListener(@NonNull LocationListener locationListener)`

But the same method in the `IndoorPositioning` (Which should be named IndoorPositioningProvider - i guess) are nullable.

`registerLocationListener(LocationListener locationListener)`

Can we follow the same convention in both methods ?

Steppschuh, 2018-06-18:
That's true, but won't change anytime soon. It's because the `@NonNull` annotation is part of `android.support.annotation` and thus not available in the Java core library.

closed 2018-06-18 by Steppschuh

hadiidbouk, 2018-06-19:
But isn't the Java 8 contains a `@NonNull` annotation ?

Check [this](https://blogs.oracle.com/java-platform-group/java-8s-new-type-annotations)

Steppschuh, 2018-06-19:
Thanks for the heads up. We'll see if and how we can combine these annotations.