neXenio/BLE-Indoor-Positioning #94: App crash in updateLocation method

hadiidbouk, 2018-06-20:
Hello,

I found that the App crash when the number of my beacons is 4 in `updateLocation()` method inside the `IndoorPositioning` class

`No interface method sort(Ljava/util/Comparator;)V in class Ljava/util/List; or its super classes (declaration of 'java.util.List' appears in /system/framework/core-libart.jar)`

1 - 
[image]

2 -
[image]

3 -
[image]

hadiidbouk, 2018-06-20:
After searching a little bit i found that it's not recommended to use ArrayList#sort if the app targeting Api Level > 25.

Instead we should use `Collections.sort(list, comparator)`.

```
Implementation Note:
This implementation defers to the List.sort(Comparator) method using the specified list and a null comparator. Do not call this method from List.sort() since that can lead to infinite recursion. Apps targeting APIs <= 25 observe backwards compatibility behavior where this method was implemented on top of List.toArray(), ListIterator.next() and ListIterator.set(Object). 
```

check [this](https://developer.android.com/reference/java/util/Collections#sort(java.util.List%3CT%3E)) and [this](https://developer.android.com/reference/java/util/ArrayList#sort(java.util.Comparator%3C?%20super%20E%3E)).

I sent a [PR](https://github.com/neXenio/BLE-Indoor-Positioning/pull/95) to fix this bug.

Steppschuh, 2018-06-20:
I find it strange that we didn't ran into this issue, as we're also targeting API level 26 and have > 3 beacons in range all the time. Anyway, glad you figured out that using `Collections.sort(list, comparator)` fixed this issue for you.

closed 2018-06-20 by Steppschuh