neXenio/BLE-Indoor-Positioning #97: Add getter and setter for ROOT_MEAN_SQUARE_THRESHOLD

hadiidbouk, 2018-06-21:
The current code inside `updateLocation()` is like this 

```
 if (multilateration.getRMS() < ROOT_MEAN_SQUARE_THRESHOLD_STRICT) {
            locationPredictor.addLocation(location);
            onLocationUpdated(getMeanLocation(2, TimeUnit.SECONDS));
        }

```
The user should choose between threshold : `ROOT_MEAN_SQUARE_THRESHOLD_STRICT`, `ROOT_MEAN_SQUARE_THRESHOLD_MEDIUM`, `ROOT_MEAN_SQUARE_THRESHOLD_LIGHT`

Check the PR [here](https://github.com/neXenio/BLE-Indoor-Positioning/pull/96)

Steppschuh, 2018-06-21:
Out of curiosity, what values do you find being a better fit than the current default value?

hadiidbouk, 2018-06-21:
None, but i am trying to test my current beacons Rssi signal with my custom walking detector ( if the user isn't walking don't show a new location ), the fact that i have added a -16dBm as Tx power, so if i choose the `ROOT_MEAN_SQUARE_THRESHOLD_STRICT` i am not getting any location.

 You can see here in the image, the ROOT_MEAN_SQUARE_THRESHOLD_STRICT will definitely not working, so at least it should be customized for testing purpose.

I still don't know how to find the best TxPower based on the area and how far the beacon is from the other and how many beacons i need in a floor..

[image]

PR #96 (2018-06-21): Add getter and setter for ROOT_MEAN_SQUARE_THRESHOLD
Closes #97