neXenio/BLE-Indoor-Positioning #112: Location inside a room

amit2103, 2018-07-28:
Hey,
Thanks for the awesoem work.
I plan on using some estimote proximity beacons inside some rooms.
Now I want to detect if  a person went inside a room or outside.
Is there any way to say if a person is inside a room or outside using this app or even doing  code modification ? Any help will be much appreciated.

hadiidbouk, 2018-07-28:
Hello,

I think this is depending on which map provider you use,
In my company, we use Esri maps, the data team draw "the walking area" part of the map, and I can check if the location detected is inside this walking area or not.

Also, I can think with an idea is to put some beacons around the room, each one of these beacons should have a (0,0) coordinates.
Then you can calculate the distance between your device in meters if the new location is like (-3,5) for example that's mean that you are out of the room by 3 meters to the left.

But sure you should use fake coordinates based on meters, we have tried that logic and it works, we put the map on the (0,0) coordinates (it will be somewhere on the sea).

Please note that if you put the first beacon on (0,0) and the second beacons on (5,5) you should test if the new location detected has an x < than 5.

amit2103, 2018-07-31:
> Hey,
> a very simple solution would be to place some beacons inside the room and
> some outside. Then you calculate the distances to all your beacons and
> check if the closest beacon is one that is inside the room. I think
> Estimote provides this functionality out of the box with their SDK. You
> should contact their sales or support team regarding this, I guess.
>
> This library is for more advanced indoor positioning, though. You can use
> it to calculate geo coordinates based on the received advertising frames
> from (any) Bluetooth beacons, as long as you know the beacon locations.

Hey I want to use this library to calculate geo coordinates as I said. But
i also want to point out where the person is in a map. hence the question.
We are not using a mapping provider yet

Steppschuh, 2018-08-01:
If you have the geo-coordinates, you can plot them on any map you like (using a [projection](https://en.wikipedia.org/wiki/Map_projection)). You could use things like Google Maps or implement it on your own. This library already contains some utility classes for that, e.g. the `SphericalMercatorProjection`. You can find them in the `com.nexenio.bleindoorpositioning.location.projection` package.

I started working on a map view for the demo app that renders the device location on a given floor plan, which sounds like exactly what you want. You can find the work in progress [here](https://github.com/neXenio/BLE-Indoor-Positioning/pull/90).

Steppschuh, 2018-08-05:
No, but I'm sure there won't be any activity from our side on this issue before the 21st of this month. Feel free to fork and continue working on it, though.

Steppschuh, 2018-10-30:
Merged in #90

closed 2018-10-30 by Steppschuh