neXenio/BLE-Indoor-Positioning #114: Blueup advertising packet length

hadiidbouk, 2018-08-02:
Hello, 

In your IBeaconTest class, you added the blueup frame data, this frame data is smaller than the kontakt frame data : 

```
 public final static byte[] KONTAKT_FRAME_DATA = new byte[]{2, 1, 6, 26, -1, 76, 0, 2, 21, -111, 20, -42, 26, 103, -47, 17, -24, -83, -64, -6, 122, -32, 27, -66, -68, 100, -16, -113, 90, -59, 8, 9, 75, 111, 110, 116, 97, 107, 116, 2, 10, 4, 10, 22, 13, -48, 119, 99, 74, 111, 52, 50, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0};

 public final static byte[] BLUEUP_FRAME_DATA = new byte[]{2, 1, 6, 26, -1, 76, 0, 2, 21, 3, 37, 63, -35, 85, -53, 68, -62, -95, -21, -128, -56, 53, 95, -126, -111, 0, 1, 0, 2, -54};

```

My Question is, if i want to get the uuid bytes from this array, I will go and pick the bytes from index 9 to index 24+9 like you did in your code.

This logic will let me get a wrong uuid bytes, for blueup i am getting 13 bytes and for kontakt i am getting 16 bytes which is the correct number by the iBeacon advertising packet standards.

How i suppose to get the uuid from only this 13 bytes ?

Steppschuh, 2018-08-02:
I'm not sure if I understand the issue. You can get the UUID by calling `getProximityUuid()` on the advertising packet, just as we did in the `IBeaconTest`:

```java
@Test
public void getProximityUuid() {
    assertEquals(UUID.fromString("9114d61a-67d1-11e8-adc0-fa7ae01bbebc"), kontaktAdvertsingPacket.getProximityUuid());
    assertEquals(UUID.fromString("03253fdd-55cb-44c2-a1eb-80c8355f8291"), blueupAdvertsingPacket.getProximityUuid());
}
```

Internally it uses the `AdvertisingPacketUtil` to extract the UUID from the given byte array:

```java
public static UUID toUuid(byte[] bytes) {
    ByteBuffer bb = ByteBuffer.wrap(bytes);
    return new UUID(bb.getLong(), bb.getLong());
}
```

hadiidbouk, 2018-08-06:
Nevermind, it was a silly problem.

closed 2018-08-06 by hadiidbouk