neXenio/BLE-Indoor-Positioning #117: AdvertisingPacketUtil:getAdvertisingPacketsBetween() Return a wrong list

hadiidbouk, 2018-08-06:
Hello,

According to the description, the method below "_**Returns an ArrayList of AdvertisingPackets that have been received in the specified time range.**_"
```
public static <P extends AdvertisingPacket> ArrayList<P> getAdvertisingPacketsBetween(final ArrayList<P> advertisingPackets, long startTimestamp, long endTimestamp) {
          ...
    }
```
I am testing this method right now, Here is my input data : 

```
long startTimestamp = 1533538629;
long endTimestamp = 1533538899;
         
ArrayList<AdvertisingPacket> advertisingPackets = new ArrayList<AdvertisingPacket>();
advertisingPackets.add(new AdvertisingPacket(1, 1533538511));
advertisingPackets.add(new AdvertisingPacket(2, 1533538629));
advertisingPackets.add(new AdvertisingPacket(3, 1533538686));
advertisingPackets.add(new AdvertisingPacket(4, 1533538737));
advertisingPackets.add(new AdvertisingPacket(5, 1533538772));
advertisingPackets.add(new AdvertisingPacket(6, 1533538805));
advertisingPackets.add(new AdvertisingPacket(7, 1533538849));
advertisingPackets.add(new AdvertisingPacket(8, 1533538865));
advertisingPackets.add(new AdvertisingPacket(9, 1533538884));
advertisingPackets.add(new AdvertisingPacket(10, 1533538899));
advertisingPackets.add(new AdvertisingPacket(11, 1533538912));
advertisingPackets.add(new AdvertisingPacket(12, 1533539012));
```
After calling `getAdvertisingPacketsBetween` method with the input above the result returned is 
[ 2, 3, 4, 5, 6, 7, 8, 9], Why we aren't taking the index number 10 with the array ? 
I think it's a bug, Every endTimestamp chosen > 6 ( which it's the middle of the list), the last index is skipped.

To Solve the issue above, we should add 1 to the endIndex calculated when 
`endTimestamp >= midstAdvertisingPacket.getTimestamp()
`    ```
    // find the index of the last advertising packet with a timestamp
        // smaller than the specified endTimestamp
        int endIndex = advertisingPackets.size() - 1;
        if (endTimestamp < latestAdvertisingPacket.getTimestamp()) {
            // figure out if the end timestamp is before or after the midst advertising packet
            ListIterator<P> listIterator;
            if (endTimestamp < midstAdvertisingPacket.getTimestamp()) {
                // end timestamp is in the first half of advertising packets
                // start iterating from the beginning
                listIterator = advertisingPackets.listIterator(startIndex);
                while (listIterator.hasNext()) {
                    if (listIterator.next().getTimestamp() >= endTimestamp) {
                        endIndex = listIterator.previousIndex();
                        break;
                    }
                }
            } else {
                // end timestamp is in the second half of advertising packets
                // start iterating from the end
                listIterator = advertisingPackets.listIterator(advertisingPackets.size());
                while (listIterator.hasPrevious()) {
                    if (listIterator.previous().getTimestamp() < endTimestamp) {
                        endIndex = listIterator.nextIndex() **+ 1**;
                        break;
                    }
                }
            }
        }
```

hadiidbouk, 2018-08-07:
I have run this test over and over again,
It seems that the code below never executed, I set the breakpoint in the else statement where I made the changes.

```
} else {
    // end timestamp is in the second half of advertising packets
    // start iterating from the end
       listIterator = advertisingPackets.listIterator(advertisingPackets.size());
       while (listIterator.hasPrevious()) {
             if (listIterator.previous().getTimestamp() < endTimestamp) {
                        endIndex = listIterator.nextIndex() + 1;  <------ HERE
                        break;
                    }
              }
         }
```

marvinmirtschin, 2018-08-08:
You are correct, i originally used other values for `numberOfPackets` (greater 2) which will result in the described behaviour while executing this code block.

My test revealed a bug in our code.
```
        // find the index of the last advertising packet with a timestamp
        // smaller than the specified endTimestamp
        int endIndex = advertisingPackets.size() - 1;
        if (endTimestamp < latestAdvertisingPacket.getTimestamp()) {
            // figure out if the end timestamp is before or after the midst advertising packet
            ListIterator<P> listIterator;
            if (endTimestamp < midstAdvertisingPacket.getTimestamp()) {
                // end timestamp is in the first half of advertising packets
                // start iterating from the beginning
                listIterator = advertisingPackets.listIterator(startIndex);
                while (listIterator.hasNext()) {
                    if (listIterator.next().getTimestamp() >= endTimestamp) {
                        endIndex = listIterator.previousIndex();     // ---> a)
                        break;
                    }
                }
            } else {
                // end timestamp is in the second half of advertising packets
                // start iterating from the end
                listIterator = advertisingPackets.listIterator(advertisingPackets.size());
                while (listIterator.hasPrevious()) {
                    if (listIterator.previous().getTimestamp() < endTimestamp) {
                        endIndex = listIterator.nextIndex();            // ---> b)
                        break;
                    }
                }
            }
        }

        return new ArrayList<>(advertisingPackets.subList(startIndex, endIndex + 1));
```
In the line `a)` the index of the first incorrect packet will be found, while in line `b)` the index of the last correct packet will be found. The problem is, that we did not adjust line `b)` but line `c)` resulting in adjusting both cases, `a)` and `b)`.

To get back to the original question if you would like to make `endTimestamp` inclusive you must adjust the line above `a)` from `>=` to `>` and the line above `b)` from `<` to `<=`.

hadiidbouk, 2018-08-08:
Where is the line c) ?

marvinmirtschin, 2018-10-09:
Fixed in #122

closed 2018-10-09 by marvinmirtschin