neXenio/BLE-Indoor-Positioning #120: getAdvertisingPacketBetween() in AdvertisingPacketUtil returns incorrect list

marvinmirtschin, 2018-08-08:
```
        // find the index of the last advertising packet with a timestamp
        // smaller than the specified endTimestamp
        int endIndex = advertisingPackets.size() - 1;
        if (endTimestamp < latestAdvertisingPacket.getTimestamp()) {
            // figure out if the end timestamp is before or after the midst advertising packet
            ListIterator<P> listIterator;
            if (endTimestamp < midstAdvertisingPacket.getTimestamp()) {
                // end timestamp is in the first half of advertising packets
                // start iterating from the beginning
                listIterator = advertisingPackets.listIterator(startIndex);
                while (listIterator.hasNext()) {
                    if (listIterator.next().getTimestamp() >= endTimestamp) {
                        endIndex = listIterator.previousIndex();     // ---> a)
                        break;
                    }
                }
            } else {
                // end timestamp is in the second half of advertising packets
                // start iterating from the end
                listIterator = advertisingPackets.listIterator(advertisingPackets.size());
                while (listIterator.hasPrevious()) {
                    if (listIterator.previous().getTimestamp() < endTimestamp) {
                        endIndex = listIterator.nextIndex();            // ---> b)
                        break;
                    }
                }
            }
        }

        return new ArrayList<>(advertisingPackets.subList(startIndex, endIndex + 1));
```

In the line a) the index of the first incorrect packet will be found, while in line b) the index of the last correct packet will be found. The problem is, that we did not adjust line b) but line c) resulting in adjusting both cases, a) and b).

hadiidbouk, 2018-08-11:
Any updates from this?

Steppschuh, 2018-08-16:
Please excuse the delay, we're on a tight schedule this week and won't be able to work on this repo before the 22nd.

The real issue here is that the method in question is lacking unit tests to assert that it behaves as expected. We'll add tests as soon as time permits.

commit 1c087a52 (2018-08-17): Merge pull request #122 from neXenio/0120-fix-getAdvertisingPacketsBetween

fixes #120; adds new tests for AdvertisingUtil.getAdvertisingPacketsBetween
commit 67906e60 (2018-08-17): fixes #120; adds new tests for AdvertisingUtil.getAdvertisingPacketsBetween
closed 2018-08-20 by Steppschuh