neXenio/BLE-Indoor-Positioning #123: Error: Program type already present

Khizrum, 2018-10-01:
I have been successful in building the project with no errors in java or layout, but when i'm running the code, so I'm getting the error... Error: Program type already present: com.nexenio.bleindoorpositioning.ble.advertising.AdvertisingPacket..

I'm stuck here and im unable to test this application. Any help will be immensely appreciated..
[image]

Steppschuh, 2018-10-02:
Hey, thanks for checking out this project!

You are declaring multiple dependencies on this project in your `build.gradle`, you should remove all but one of them.

You should use `implementation project(':BLE Indoor Positioning')` when you are working on the [demo app](https://github.com/neXenio/BLE-Indoor-Positioning/tree/dev/app) module. This will always re-compile the library module and you will have changes made in the library module instantly available in the app module.

You should use `implementation 'com.github.neXenio:BLE-Indoor-Positioning:dev-SNAPSHOT` if you are working on your own project and want the latest changes of branch (e.g. `dev`) from this GitHub project. Check out [JitPack](https://jitpack.io/#neXenio/BLE-Indoor-Positioning/) for details.

To get reproducible builds, you should avoid `-SNAPSHOT` dependencies. You can still use JitPack for that, just replace `dev-SNAPSHOT` with a release version name (e.g. `0.3.0`) or a commit hash (e.g. `a87fe8b`).

closed 2018-10-02 by Steppschuh

Khizrum, 2018-10-20:
Dear Mr. Schultz,
I'm extremely grateful that you responded, unfortunately I couldn't see your mail sooner because your mails were sent in my spam box that I recovered today. I implemented your suggestions and they worked, but now I'm facing new errors. The errors state that they aren't able to resolve setElevation and getTimeStamp methods. I'm attaching the screenshots below, I hope you'll respond soon..
kind regards,Khizrum

Sent from Yahoo Mail. Get the app

Khizrum, 2018-10-20:
When I used the project (':BLE Indoor Positioning') dependency then the screen that I have attached popped up.... I switched on my beacon but nothing really happened... What should i do?

Steppschuh, 2018-10-22:
Please create new issues for problems that are not related to this one. Also, I can't see any attachments because you responded to the GitHub notification emails instead of using the website.