neXenio/BLE-Indoor-Positioning #144: To implement in India

Manigandan21, 2018-11-27:
Does this application need any major change to test in India? I can see a blank page for Map, Radar, and Chart.

My Current location is **Last known location set to Latitude: 13.092544 Longitude: 80.1626726**

I am using IBeacon Minor value of 3 beacons. Minor values set to 100, 200 and 300.

Sample part:

case 100: {
                beaconLocation.setLatitude(13.092441779956898);
                beaconLocation.setLongitude(80.16278815262923);
                beaconLocation.setAltitude(20);
                break;
            }
Now I can see that application is not loading Map image . 

Also when I check the log for the following I can see infinity value.
fromLocation.getLatitude(), fromLocation.getLongitude(), fromLocation.getAltitude(),toLocation.getLatitude(), toLocation.getLongitude(), toLocation.getAltitude()

**Infinity::-Infinity,0.0,-Infinity,Infinity,0.0**

Can you please help.

Steppschuh, 2018-11-27:
The chart view shows all beacons that match the current [UUID filter](https://github.com/neXenio/BLE-Indoor-Positioning/blob/0ce40a7db42775dc2683cd616280dac443bd4168/app/src/main/java/com/nexenio/bleindoorpositioningdemo/ui/beaconview/BeaconViewFragment.java#L41). You can either adjust that filter to accept your beacons, or configure your beacons to advertise frames that match the current filter implementation (IBeacon frames with UUID `03253fdd-55cb-44c2-a1eb-80c8355f8291`).

The map and radar views will only show something meaningful if you have a `LocationProvider` assigned to each of your beacons. The dummy app currently does that in the [`BluetoothClient`](https://github.com/neXenio/BLE-Indoor-Positioning/blob/0ce40a7db42775dc2683cd616280dac443bd4168/app/src/main/java/com/nexenio/bleindoorpositioningdemo/bluetooth/BluetoothClient.java#L136).

To get a background image in the map view, adjust the [BeaconMapFragment](https://github.com/neXenio/BLE-Indoor-Positioning/blob/0ce40a7db42775dc2683cd616280dac443bd4168/app/src/main/java/com/nexenio/bleindoorpositioningdemo/ui/beaconview/map/BeaconMapFragment.java#L82) and replace the [image drawable](https://github.com/neXenio/BLE-Indoor-Positioning/blob/0ce40a7db42775dc2683cd616280dac443bd4168/app/src/main/res/drawable-nodpi/map_view_background.png) with a floor plan of your location. Calculating the reference locations is quite complex, though. You can read about that in the [wiki](https://github.com/neXenio/BLE-Indoor-Positioning/wiki/Calculating-Geo-Coordinates-of-Beacons).

closed 2018-11-27 by Steppschuh

Manigandan21, 2018-11-28:
Sorry I failed to add the code in insert code. Please check now.

As per the LOG the execution jumps to else if part. So I can't see the user location in the Map view.

Steppschuh, 2018-11-28:
And how is that related beacons not showing up in the view?

The `IndoorPositioning` singleton uses beacons that have a `LocationProvider` to calculate the geolocation of the device that received the beacon advertising frames.

The fact that the code always jumps to the `else if` part means that you're not receiving location updates from `IndoorPositioning`, most likely because you didn't configure location providers for your beacons (or the provided locations are so wrong that the multilateration is impossible).