neXenio/BLE-Indoor-Positioning #148: ClassCastException: Eddystone cannot be cast to IBeacon

Steppschuh, 2019-03-12:
```java
java.lang.ClassCastException: com.nexenio.bleindoorpositioning.ble.beacon.Eddystone cannot be cast to com.nexenio.bleindoorpositioning.ble.beacon.IBeacon
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.bleindoorpositioning.ble.beacon.filter.IBeaconFilter.matches(IBeaconFilter.java:14)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.bleindoorpositioning.ble.beacon.FilteredBeaconUpdateListener.onBeaconUpdated(FilteredBeaconUpdateListener.java:24)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.bleindoorpositioning.ble.beacon.BeaconManager.notifyBeaconUpdateListeners(BeaconManager.java:108)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.bleindoorpositioning.ble.beacon.BeaconManager.processAdvertisingPacket(BeaconManager.java:83)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.bleindoorpositioning.ble.beacon.BeaconManager.processAdvertisingData(BeaconManager.java:59)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.seamlessauthentication.internal.accesscontrol.BaseAccessControlDetector.lambda$processBeaconAdvertisingPackets$3(BaseAccessControlDetector.java:150)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at com.nexenio.seamlessauthentication.internal.accesscontrol.-$$Lambda$BaseAccessControlDetector$zt5Rf25lr9DLmQmilGX7pfTYro8.subscribe(Unknown Source:2)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at io.reactivex.internal.operators.maybe.MaybeCreate.subscribeActual(MaybeCreate.java:45)
2019-03-12 15:58:50.113 15919-15919/com.nexenio.seamlessauthentication W/System.err:     at io.reactivex.Maybe.subscribe(Maybe.java:4262)
```

NishantMeena, 2019-05-09:
hiii i am using your sample in which the beconmapfragment not working actually one exception occure which i mention below,

java.lang.ClassCastException: com.nexenio.bleindoorpositioning.ble.beacon.Eddystone cannot be cast to com.nexenio.bleindoorpositioning.ble.beacon.IBeacon

So please help me to resolve my issue thanks in advance.

Steppschuh, 2019-05-09:
Currently you need to make sure that you only pass `IBeacon` instances to an `IBeaconFilter` (although the interface may allow generic `Beacon` instances). If you pass something else (e.g. an `Eddistone`), you will get the exception from above.

You could add a simple `instanceof` check for now.

NishantMeena, 2019-05-09:
i am just download youor source code and import it to my android studio,i have 4 beacons and i also crerate my own image for beaconmap background . and i set my top left corner latlong and right bouttom latlong.can you please explain to me how can i do that in your sample map fragment.

Steppschuh, 2019-05-09:
> please explain to me how can i do that in your sample map fragment.

That is not related to this issue. Please refer to #144

NishantMeena, 2019-05-09:
i just need to know if i want to set my indoor map image and beacon's in your sample app so which type of changes required.

PR #156 (2019-08-15): 0148 classcastexception
Resolves #148 

Breaking changes:

- `BeaconFilter` requires a new method: `boolean canMatch(Beacon beacon)`
closed 2019-09-10 by Steppschuh