neXenio/BLE-Indoor-Positioning #149: ibeacons are not getting detected which are stimulated using beacon simulator

Keyurkariya, 2019-03-13:
There are mac id's detected of the bluetooth but does not detect the beacons around and the advertising packet is null.

Steppschuh, 2019-03-13:
That most likely means that your simulation is not advertising valid iBeacon frames.

You can check the requirements in [IBeaconAdvertisingPacket.java](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/ble/advertising/IBeaconAdvertisingPacket.java#L70)

Please post the exact advertising data that you are sending.

Keyurkariya, 2019-03-15:
`2019-03-15 12:55:56.929 25539-25539/com.nexenio.bleindoorpositioning D/BluetoothClient: in Process Scan Result
2019-03-15 12:55:56.930 25539-25539/com.nexenio.bleindoorpositioning D/BluetoothClient: the macaddress is 38:4B:D4:63:E5:96
2019-03-15 12:55:56.933 25539-25539/com.nexenio.bleindoorpositioning D/BluetoothClient: the scan result is ScanResult{bleDevice=RxBleDeviceImpl{bluetoothDevice=null(38:4B:D4:63:E5:96)}, rssi=-84, timestampNanos=48765202214458, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=com.polidea.rxandroidble.internal.scan.ScanRecordImplNativeWrapper@f64ce4}
2019-03-15 12:55:56.933 25539-25539/com.nexenio.bleindoorpositioning D/BluetoothClient: advertising packet isnull`

This is the logcat and i am stimulating ibeacon using the Beacon Stimulator Application On Play Store.

Steppschuh, 2019-03-15:
With exact advertising data I meant the [byte array](https://developer.android.com/reference/android/bluetooth/le/ScanRecord.html#getBytes()) that is part of the `scanRecord`. There are very specific rules for how the advertising frames are structured:

[image]

closed 2019-03-15 by Keyurkariya