neXenio/BLE-Indoor-Positioning #151: Unable to detect Eddystone Beacon Stimulated by Beacon Stimulator

Keyurkariya, 2019-03-15:
I am trying to see eddystone beacons where i can just set up namespace and instance, what am i supposed to make change to my EXPECTED_EDDYSTONE_UUID in EddystoneAdvertisingPacket.java

Keyurkariya, 2019-03-15:
The app does not even show anything on the radar neither on the chart.

Steppschuh, 2019-03-15:
I guess you should use `IndoorPositioning.getInstance().setIndoorPositioningBeaconFilter(...)` with a filter that matches your beacons. To get it to work in the demo app, adjust the `uuidFilter` in `BeaconViewFragment`

allen31638, 2019-04-10:
I have tried to modify ("acfd065e-c3c0-11e3-9bbe-1a514932ac01") and the chart will be displayed.
But the radar still can't display.
Request assistance.

Steppschuh, 2019-04-10:
If you see beacons in the chart but not in the map or radar, you probably have not specified a `LocationProvider` for your beacons. The chart can render the RSSIs and distances, but the other views require geo locations in order to render.

You can read more about that in #150

allen31638, 2019-04-16:
@Steppschuh 
Sorry, I still don't understand this. I went to see the #150 article or I don't know where to change.
There is still no correction radar.Please ask for teaching, thank you
        Location firstReferenceLocation = new Location(25.126090, 121.500060);
        Location secondReferenceLocation =new Location(25.126095, 121.500065);

        Point firstReferencePoint = new Point(953, 1830);
        Point secondReferencePoint = new Point(1926, 1830);

Steppschuh, 2019-04-16:
Reference points are for the map view only, that's not related to the radar.

There are code snippets in the linked issue, where `setLocationProvider(...)` is used. That's what you need to do for your beacons, obviously with adjusted geo locations that match your setup. There's a guide for calculating the geo locations in the [wiki](https://github.com/neXenio/BLE-Indoor-Positioning/wiki/Calculating-Geo-Coordinates-of-Beacons).

closed 2019-09-16 by Steppschuh