neXenio/BLE-Indoor-Positioning #152: Help ! How exactly is the UI for map is working.

Keyurkariya, 2019-03-19:
I am trying to build an android application which will map all the beacons along with the location of the person. I am scanning the beacons with AltBeacon Library and have successfully obtained the X,Y coordinates of the person. But i have no clue as to how to map this and show a live visual representation of this.

So currently i have the X,Y coordinates of all beacons which are not exactly lat long but some random values like (0,0) (1,1) etc. And i am finding the distance using altbeacon library from the nearest beacons and calculating the centroid and also getting a X,Y coordinate. Do help to build a UI representation of beacons and the person on an indoor map layout.

Steppschuh, 2019-03-19:
Drawing a map on canvas is not an easy task. You will have to deal with projections and quite a bit of math. Please refer to PR #90 to get an idea for how the map view is implemented.

I guess you'll need to find a way to get the actual geo-coordinates from the beacons you are using.

Keyurkariya, 2019-03-19:
I can get the geo-coordinates using GPS Logger, prior to running the application and which relatively will calculate the Coordinates for User using Trilateration. Will this work ?

Steppschuh, 2019-03-19:
Doing that is exactly the point of this library.

Keyurkariya, 2019-03-19:
But the scanning of BLE Beacons and finding the distance is very complexly done here

Steppschuh, 2019-03-19:
Scanning is not part of this library, because there are sophisticated libraries available for that (as mentioned in the readme). Finding the distance is as complex as calling `beacon.getDistance()` (as mentioned in the readme).

Keyurkariya, 2019-03-22:
Okay, so can you guide me how to find the (x,y) coordinate of the user, after i found all the beacons around me along with their distances and are stored in a hashmap. Also where exactly to store the Lat longs of the beacons attached in the area.

closed 2019-09-16 by Steppschuh