neXenio/BLE-Indoor-Positioning #162: Eddystone cannot be cast to IBeacon

shakib1377, 2019-09-07:
Hi
I just cloned this source and changed INDOOR_POSITIONING_UUID to uuid of my own beacons and installed that on my phone. in Chart fragment I can see beacons and their RSSI on the chart but once i click on Map or Radar button the app will crash with this error:

> java.lang.ClassCastException: com.nexenio.bleindoorpositioning.ble.beacon.Eddystone cannot be cast to com.nexenio.bleindoorpositioning.ble.beacon.IBeacon
        com.nexenio.bleindoorpositioning.ble.beacon.filter.IBeaconFilter.matches(IBeaconFilter.java:15)
        com.nexenio.bleindoorpositioningdemo.ui.beaconview.BeaconViewFragment.getBeacons(BeaconViewFragment.java:136)
        com.nexenio.bleindoorpositioningdemo.ui.beaconview.radar.BeaconRadarFragment.onCreateView(BeaconRadarFragment.java:110)

so anyone knows how can I fix this?
any help will be much appreciated.

Steppschuh, 2019-09-10:
Hey,
that sounds related to #148, which has recently been fixed in #156 and released in #160.

Keep in mind that you can't use an `IBeaconFilter` to filter `Eddystone` beacon instances, as it's intended to work with `IBeacon` instances only. You'd need to create a new `BeaconFilter` (e.g. by extending `GenericBeaconFilter `) in order to do that. You can find the filters [here](https://github.com/neXenio/BLE-Indoor-Positioning/tree/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/ble/beacon/filter)

shakib1377, 2019-09-15:
Thank you so much :) problem solved

closed 2019-09-16 by Steppschuh