neXenio/BLE-Indoor-Positioning #163: Radar and Map fragments doesn't show anything

shakib1377, 2019-09-15:
Hi everyone
I just build the cloned project and after just simply changed the uuidFilter value from 

> "acfd065e-c3c0-11e3-9bbe-1a514932ac01"

to 

> "23a01af0-232a-4518-9c0e-323fb773f5ef"

which are my beacons UUID , after that I installed that on my phone and when I'm clicking in chart everything is fine and the app show me a RSSI chart of my beacons but when I'm clicking on **Radar** or **Map** my beacons **won't be shown** (In radar I'm just seeing a orange circle on center but any of beacons aren't shown and in Map a square is shown in left top of screen and nothing else)
**My question is that** is there any configuration that must be done for Radar and Map fragments in order to show the radar and map of the beacons?
is there anyone who can help me with this?
any help will be much appreciated.

Steppschuh, 2019-09-16:
Hey,
make sure that you also have a valid `LocationProvider` set for your beacons. Please refer to #144 for details.

closed 2019-09-16 by Steppschuh

jasirjasir, 2020-07-03:
comment this line in BeaconRadarFragment.java to disable UUID filter
beaconFilters.add(uuidFilter);