neXenio/BLE-Indoor-Positioning #168: How to implement Bluetooth Scanning?

KyleAdo, 2020-06-13:
I'm currently trying this out, but the only part of the documentation I'm struggling with is the Bluetooth scanning part which I'm not really understanding. 

I'm not sure if it's already built into the code, or if it's something that I need to implement entirely by myself? I have 4 Ibeacons and I've inputted the correct longitude/latitude coordinates for the 4 of them in the IbeaconLocation Provider method in the Bluetooth Client class. However when I run the app through my phone it's not picking up anything, it's completely blank on radar/chart/map. 

I've got the UUID, minor and major values for each of my beacons, is there anywhere in the code where I need to input them? Or is this something I'll have to do myself? Because that's the only issue I'm having so far is actually connecting to the beacons themselves.

Steppschuh, 2020-06-15:
The scanning is not part of the library. However, you can check out the sample application module which is also part of this repository. It implements the scanning and shows how the library can be used.

Also refer to #144 if no beacons are showing up.

KyleAdo, 2020-06-17:
Thank you, I can see the examples in the BluetoothClient class, but they look like they're already implemented and should be working, so I'm confused on what I have to do in order to start the scanning, and obtaining the beacon API which will then be sent to the ScanResult method you built. 

Forgive me I know this isn't your job and it's something people have to do themselves, but is there any pointers you can give me on what to do to get the scans working correctly? I'm trying to use the Bluetooth Adapter that you recommended but I'm not sure where to place the LeScan methods when there's already a scanning method that should do the job anyway. 

If there's nothing else you can offer it's completely fine I just thought I would enquire, I'm sure I'll get there eventually :)

Steppschuh, 2020-06-17:
The readme contains a section about that, including example code: [Bluetooth Scanning](https://github.com/neXenio/BLE-Indoor-Positioning#bluetooth-scanning).

There really isn't anything I could add to that. You need to pass advertising data to the library (using `BeaconManager.processAdvertisingData(...)`). How you get the data depends on your implementation, examples are linked in the readme and implemented in the demo app.

KyleAdo, 2020-06-18:
Okay thank you very much for your help! Appreciate it, I'll close the topic :)

closed 2020-06-18 by KyleAdo