neXenio/BLE-Indoor-Positioning #173: Map background image moves out of position

KyleAdo, 2020-07-21:
Hello again, really enjoying the implementation so far, however, just having an issue with the Map section. I've currently set up a test scenario in my home and set up 4 beacons around my house. The only part that's not working as anticipated is the Map section. I think this is mostly because RSSI values aren't very strong, and the coordinates have me in many different places every 5 seconds, but anyway, this is what it looks like. 

[image]

The beacons are showing correctly, and they're supposed to be in all the 4 corners of my house, and I think I've got the correct geocoordinates for each of them. I'm using a Samsung Galaxy J330, I'm not sure if it's signal strength that's messing it up because I'm clearly not outside when I'm running it, sometimes it has the correct the position, then sometimes it has me outside the house as it does there and as soon as that happens the image is completely ruined. 

Here are the reference location/points (I've blocked out some of my coordinates for privacy reasons) 

```
Location firstReferenceLocation = new Location(54.5*****, -1.1*****);
Location secondReferenceLocation = new Location(54.5*****, -1.1*****);

Point firstReferencePoint = new Point(500, 240);
Point secondReferencePoint = new Point(1000, 500);

```

Also, I'll add I'm not that familiar with the x and y reference points, so I just played around with them until I got a good enough fit, however, none of that matters since it keeps resizing and moving out of position anyway. 

Sorry to make it a big post, any answer or advice is appreciated, I'm just not sure if this is normal or if it's something to do with my phone obtaining weak RSSI values. I have noticed when I run the application and read the AndroidLocationProvider values in the console don't produce the exact same coordinates each 5 seconds, there is quite a decent leap between each, is that normal?

Steppschuh, 2020-07-22:
Hey, you seem to have two distinct issues:

First is the mapping of the beacons to your background image (reference locations and reference points). However, the map shouldn't move away from the beacons at runtime. That might be a bug in the demo app.

Second is the indoor positioning. Looking at your floor plan, this can't work well in your setup - even if you beacons would be calibrated properly. The multilateration works be estimating distances to each beacon (visualized by the semi-transparent blue circle around each beacon) and then finding a location that has the lowest error for all estimated distances. In a perfect world, all distances would be correctly estimated and the location would be where these circles intersect. You can see on the map that the living room and kitchen beacons have very weak signals, resulting in a huge estimated distance. That is because you don't have a line of sight, the signal is blocked by walls. That's why you location can be outside, it's where the lowest error is.

You won't be able to achieve what you want using multilateration. Instead you could place one beacon on the ceiling in the center of each room and simply find the beacon with the lowest distance to detect in which room you're currently in. Alternatively, you could leave your beacons where they are and use fingerprinting (maybe also using your WiFi signal). This library won't help you with that, though.

KyleAdo, 2020-07-22:
Thank you very much for the quick response! But I see now, that's obviously why it's not working the way I'm intending it to, I'll have to come up with another plan, I'll close the thread, again thank you :)

closed 2020-07-22 by KyleAdo