neXenio/BLE-Indoor-Positioning #174: How to get pixel co-ordinate with respect to image in map fragment?

jasirjasir, 2020-07-22:
After setting Geo  coordinates to each beacon , i am able to get my current location in the map after trilateration. Is there any way to convert these  received Geo coordinates   to pixel coordinate w.r.t floor `plan ( X and Y coordinate w.r.t floor plan).

`IndoorPositioning.registerLocationListener(new LocationListener() {
    @Override
    public void onLocationUpdated(LocationProvider locationProvider, Location location) {
        //Need to convert location (latitude and longitude) data to image co-ordinate x and y
    }
});`

Steppschuh, 2020-07-28:
Sure, that's what the [CanvasProjection](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/location/projection/CanvasProjection.java) is for. You can see it being used in the [BeaconMap](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/app/src/main/java/com/nexenio/bleindoorpositioningdemo/ui/beaconview/map/BeaconMap.java#L300), e.g. when drawing the beacons. The demo app converts spherical coordinates and cartesian coordinates using the [Mercator projection](https://en.wikipedia.org/wiki/Mercator_projection).

jasirjasir, 2020-07-29:
Thanks @Steppschuh

closed 2020-07-29 by jasirjasir