neXenio/BLE-Indoor-Positioning #175: Can i use Kalman filter to remove noise data?

jasirjasir, 2020-08-12:
I have created a sperate navigation fragment in the demo app to test my trilateration algorithm independently. I am receiving x,y coordinate after trilateration and updating in the map using Canvas. But the calculated location has many noise data.  Can I use the Kalman filter available in the library package for error reduction? if Yes, How can I directly call the Kalman filter to remove noisy output data?
I have another query like is the RSSI reading in _beacon.getRssi()_ method gives the filtered RSSI value?

Steppschuh, 2020-08-18:
The signal filters are used to remove noise from the input data (i.e. the raw RSSI values), not the output (i.e. geo-coordinates).

You can specify a custom RSSI filter when requesting `beacon.getRssi(filter)`, or use the default filter with `beacon.getFilteredRssi()` and use the results for your trilateration.

You could also apply some filters to the resulting locations, but that is not implemented in this library. But keep in mind that noisy output is likely caused by noisy input. See #173 for some thoughts about that.

jasirjasir, 2020-08-18:
Thanks for the reply. I am testing my indoor navigation inside my home where so many obstacles are present like walls, tables, doors etc..I have kept a maximum of 2 beacons per room and testing my trilateration algorithm. I know the signals are influenced by thick walls. So instead, I am trying to run my trilateration code in a single room by keeping beacons on the ceiling. Hopefully, RSSI values won't be much affected by the environment at this setup. Please share your thought.
My trilateration code is accepting distance as the input. So I am reading distance using  `beacon.getDistance()` method. As per my understanding, this method uses filtered RSSI for calculating the distance. Please comment.
I am quite new to this BLE based navigation system, so maybe these questions are quite a beginner level. please consider. Thanks a lot for helping me with this project.

Steppschuh, 2020-08-18:
If you look into the [implementation](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/ble/beacon/Beacon.java#L205), you can see that there's also a `beacon.getDistance(filter)` method, which you could use with a custom filter if you don't want to use the default (which already is a `KalmanFilter`).

A note regarding your setup: As the name suggests, trilateration requires distances to three known reference points to infer the device location. This library can also make use of more reference points, hence multilateration. In both cases however, you need at least three beacons in your room, as the library will use beacons outside of you room otherwise (which you should try to avoid in order to reduce the error caused by walls). I would recommend placing a beacon in each corner of your room.

closed 2020-08-18 by Steppschuh