neXenio/BLE-Indoor-Positioning #176: How to Set Elevation and Altitude Metric in Beacon Configuration?

maroufi, 2020-08-19:
Greetings and congratulations to neXenio team for this successful project.
I have an issue in accuracy of location detection that maybe is error in my usage.
This error in accuracy of user location is 1-4 meter(s) usually.
I guess that cause of this low accuracy is error in beacon configuration or maybe in elevation and altitude parameter configuration.
I have this question that how can i set elevation and altitude in test surface and are this parameters relative to the ground level around the building?
What do you think about this location detection error?
Thank you for your attention.

Steppschuh, 2020-08-24:
If you take a look at the [Location](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/location/Location.java) class, you can find two related fields:

- `altitude`: distance to the sea level in meters
- `elevation`: relative height to the floor in meters

However, these values have no effect on the beacon distance calculation. The elevation delta is relevant when using the calculated distances for the multilateration, though. We prepared the `calculateDistanceWithoutElevationDeltaToDevice ` in [BeaconDistanceCalculator](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/location/distance/BeaconDistanceCalculator.java#L41) to offset that delta, but it's currently not used:

> Use this method to remove the elevation delta from the distance between device and beacon. Calculation based on Pythagoras to calculate distance on the floor (2D) to the beacon, if the distance is double the elevation delta. The elevation expected refers to the distance above the floor ground, rather than the altitude above sea level.

When using the [LocationDistanceCalculator](https://github.com/neXenio/BLE-Indoor-Positioning/blob/master/BLE%20Indoor%20Positioning/src/main/java/com/nexenio/bleindoorpositioning/location/distance/LocationDistanceCalculator.java), the elevation has no effect but the altitude will be taken into account.

-----------

Anyway, the elevation shouldn't have a big impact on your accuracy (unless you're in a room with a very high ceiling). Please browse other open and closed issues in this repo, there a quite a few issues related to distance estimation accuracy.

maroufi, 2020-08-24:
Thank you for your response.
I use LocationDistanceCalculator for multilateration and estimate user location (map senario).
Why should altitude effect on LocationDistanceCalculator for multilateratin while beacons and device have same altitude in a place?
Is this intended to support multi floors or it has other reason?

Steppschuh, 2020-08-25:
The `LocationDistanceCalculator` can be used to calculate the distance between `Location` instances (e.g. when calling `firstLocation.getDistanceTo(secondLocation)`). It is not used for multilateration. The altitude is relevant here because we're in a 3-dimensional space. See [this StackOverflow post](https://stackoverflow.com/questions/3694380/calculating-distance-between-two-points-using-latitude-longitude/16794680#16794680).

A `BeaconDistanceCalculator` (e.g. the `PathLossBeaconDistanceCalculator`) can be used to calculate the distance between a `Beacon` (with a known `Location`) and the device that received signals (i.e. RSSI values) from that beacon. These distances are used for multilateration. The altitude is not relevant here because id doesn't affect the measured signal strength.